duburcqa/jiminy #39: Missing unit test for Python API

duburcqa, 2020-03-02:
It would be cautious to have a basic unit test in Python checking that the Python API is working properly. The idea is just to check that it is possible to run a simulation without segfault, which may happen in practice if Eigenpy and Boost Python are not properly initialized.

matthieuvigne, 2020-03-20:
Agreed - but since we are running a simulation,  it would be nice to check the results as well :) My idea is to simulate simple linear systems with Jiminy, and make sure that the solution matches the expected analytical answer. This provides a high-level sanity check for Jiminy, that would complement the only unit test we have (which simply tests for energy conservation is a pendulum, but doesn't look at the actual answer)

duburcqa, 2020-03-21:
By the way, "duplicate" the examples in unit tests may be a good idea.

duburcqa, 2020-04-02:
@matthieuvigne Do you think we are good and that we can close this issue ?

closed 2020-04-06 by duburcqa