duburcqa/jiminy #121: Isolating pip build for Pinocchio

jcarpent, 2020-05-28:
Dear @duburcqa,

I was wondering how does it cost to isolate the pip build for Pinocchio from your current setup? I think it might be useful to do it, in order to let external users rely on Pinocchio using pip.

Thanks in advance for your help,

Justin

duburcqa, 2020-05-28:
Hi @jcarpent,

Not much, but what do you whant to do exactly ? I could deploy Pinocchio and Eigenpy as standalone packages on pip by myself, but I assume you would prefer to do it yourself ? In any case it is pretty straightforward, one just have to copy-paste `.github` folder and remove the part associated with Jiminy (after one or two minor changes).

jcarpent, 2020-05-28:
I would like to also distribute Pinocchio and EigenPy on pypy as independent packages. You've already done a great job for this support.
I would really appreciate if you can provide the skeleton, and I can ensure the rest with your support. We can imagine to host it in the stack of tasks organization if you agree and I will add you as a member. 

Is it fine?

duburcqa, 2020-05-28:
I'm pretty busy right now, but it sounds like a deal. I could add github CI doing the job directly on Pinocchio and Eigenpy repositories. It should be doable in a few hours.

jcarpent, 2020-05-28:
OK. Very nice, please proceed when you have time.
By the way, I'm trying to setup a CI for Windows using Conda, and I was wondering which compiler are you using to compile Pinocchio?

duburcqa, 2020-05-28:
I'm using Visual Studio 2019, you could have a look to this file <https://github.com/Wandercraft/jiminy/blob/master/.github/workflows/win.yml>.
Why do you want to use Conda? It is not necessary at all to build pinocchio on Windows.

jcarpent, 2020-05-28:
For installing the whole bunch of dependencies. On my side here (https://github.com/stack-of-tasks/pinocchio/blob/d921a23c99f03f3e8fba901030949dca2c0f8aa7/.github/workflows/windows-conda.yml), CMake is not able to find the compiler ClangCl, while everyone works fine on a local machine. In your case, you use `v142`. It is not working when I'm trying to compiile Pinocchio. Do you know why?

duburcqa, 2020-05-29:
Installing the dependencies without conda shouldn't be a big deal. Why to use `ClangCl` instead of `v142` on Windows ? I guess you have issues because there is spaces in your path `%programfiles(x86)%\Microsoft Visual Studio\2019\Enterprise\VC\Tools\clang++.exe`. You should get rid of them, by using the infamous windows alternative short path name. In this case, you could replace `\Microsoft Visual Studio\` by `\MICROS~1\` (6 first charatecters in uppercase, follow by ~N, N is a unique ID to distinguish folders starting with the same 6 characters.)

jcarpent, 2020-05-29:
The v142 is failing on my side dramatically. It is not able to disambiguate template signatures with Eigen objects.
In the meantime, I’ve fixed the issue which was due to cmake.

For the CI, one needs Fast feedback to get feedback on the current errors. It will be very long to wait hours for compiling dependencies while Pinocchio already has many. I would say, this is a usual and good practices to only considering the compilation of one’s projects.

closed 2020-10-13 by duburcqa