duburcqa/jiminy #219: Missing licenses in copy/pasted files

jcarpent, 2020-10-27:
Hi @duburcqa,

I really appreciate seeing Pinocchio's code to be used in external projects.

It should be nice to keep the copyright holders of the files when lines of code have been copy-pasted from original projects.
For instance, https://github.com/Wandercraft/jiminy/blob/master/python/jiminy_py/src/jiminy_py/meshcat/meshcat_visualizer.py, even if it contains some modifications.
I don't think this is the only where it occurs but as a maintener of Pinocchio, I would really appreciate to see the original credits of our work.

In addition, I see you add some features to https://github.com/Wandercraft/jiminy/blob/master/python/jiminy_py/src/jiminy_py/meshcat/meshcat_visualizer.py which can benefit to everybody. I do think you can contribute to the whole community by improving the original version here https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/pinocchio/visualize/meshcat_visualizer.py, in the same way we are trying to help you when you encounter issues or bugs.

Best regards,

Justin, on the behalf of the Pinocchio's team.

duburcqa, 2020-10-27:
Of course I will do it. I'm willing to merge those features in Pinocchio as soon as Meshcat v0.0.19 would be release because it relies on backport of master branch of Meshcat. I see no reason to maintain features on my own repository will it would be easier to maintain if merged in Pinocchio.

jcarpent, 2020-10-27:
OK. Thanks for the quick reply. I really appreciate it.
Best,

Justin

closed 2020-10-27 by jcarpent

duburcqa, 2020-10-27:
Are you OK with that: https://github.com/Wandercraft/jiminy/pull/218/commits/f616e7322ccfbbec0d13cf8c1bd7048c6aab3560 ? There should be no other copy-paste.

jcarpent, 2020-10-27:
Sounds good to me. Thanks.