duburcqa/jiminy #299: Original tutorial seems to be not working.

ashsrobbins, 2021-03-18:
Running the original tutorial .ipynb, there seem to be issues I couldn't solve. First, 
```AbstractController' object has no attribute 'set_controller_handle```
Commenting out the line trying to set the handle causes the next problem:
```TypeError: run() missing 1 required positional argument: 'v0'```
Adding a v0 variable which mimics x0 finally results in one last error:

```RuntimeError                              Traceback (most recent call last)
<ipython-input-34-90bc00b210d1> in <module>
     20 
     21 # Launch the simulation
---> 22 simulator.run(simulation_duration, x0,v0)

~/anaconda3/envs/joints/lib/python3.6/site-packages/jiminy_py/simulator.py in run(self, tf, q0, v0, is_state_theoretical, log_path, show_progress_bar)
    418                 "[{elapsed}<{remaining}]"))
    419         try:
--> 420             self.simulate(tf, q0, v0, None, is_state_theoretical)
    421         finally:  # Make sure that the progress bar is properly closed
    422             if show_progress_bar:

~/anaconda3/envs/joints/lib/python3.6/site-packages/jiminy_py/simulator.py in simulate(self, t_end, q_init, v_init, a_init, is_state_theoretical)
    384             t_end, q_init, v_init, a_init, is_state_theoretical)
    385         if return_code != jiminy.hresult_t.SUCCESS:
--> 386             raise RuntimeError("The simulation failed.")
    387 
    388     def run(self,

RuntimeError: The simulation failed.```

It doesn't get past the first step.

duburcqa, 2021-03-18:
I haven't updated the tutorial since a (very) long time. I will take care of it asap. Sorry about that.

duburcqa, 2021-03-18:
It should be working now: https://github.com/duburcqa/jiminy/blob/generic_locomotion_envs/examples/tutorial.ipynb
I will update the version on master for the next release 1.6.5, in a few days.

closed 2021-03-23 by duburcqa