duburcqa/jiminy #343: Segmentation fault when trying to use pinocchio data

vcparedesc, 2021-05-20:
Hello, I am using Jiminy tag version 1.6.17. I was trying to get the orientation and position of a particular link using a procedure I was using in tag version 1.6.1 however, I am getting segmentation faults. This is a minimal modification of the double pendulum example that reproduces my problem:

```
void computeCommand(float64_t        const & t,
                    vectorN_t        const & q,
                    vectorN_t        const & v,
                    sensorsDataMap_t const & sensorsData,
                    vectorN_t              & command)
{
    // No controller: energy should be preserved
   auto linkId = robot->pncModel_.getFrameId("PendulumArm");    
}
```

I made sure to declare "robot" as a global variable (auto robot = std::make_shared<Robot>();)

Whenever I remove the getFrameId line in computeCommand the code doesn't have any problems running, but adding the line causes the sefgaults. I am using the pip installation method for building the example.

Victor

duburcqa, 2021-05-20:
Can you provide the source file ? I'm quite surprise it is not working.

duburcqa, 2021-05-20:
Besides, do you have the trace ? You can have it using `gdb`. It would be very helpful.

duburcqa, 2021-05-20:
I tried to reproduce the issue, and it is working on my side. Here is the source file I used:
[test.txt](https://github.com/duburcqa/jiminy/files/6515664/test.txt)

vcparedesc, 2021-05-20:
I have tried it in a new computer and I got the same behavior. Both of them had previous versions of jiminy, but they were installed by source if that makes a diference.

This is what I get using gdb:
I am also uploading the file I'm using. 
```
Reading symbols from pendulum...                                                                                                                                                                                   
/home/victor/Documents/atalante_cpp_jiminy/build/core: No such file or directory.                                                                                                                                  
(gdb) run                                                                                                                                                                                                          
Starting program: /home/victor/Documents/atalante_cpp_jiminy/build/pendulum                                                                                                                                        
                                                                                                                                                                                                                   
Program received signal SIGSEGV, Segmentation fault.                                                                                                                                                               
0x00005555555711b0 in pinocchio::details::FilterFrame::operator()<double, 0> (this=0x7fffffffccb0, frame=...) at /opt/openrobots/include/pinocchio/multibody/model.hxx:30
30            { return (typeMask & frame.type) && (name == frame.name); }
(gdb) where
#0  0x00005555555711b0 in pinocchio::details::FilterFrame::operator()<double, 0> (this=0x7fffffffccb0, frame=...) at /opt/openrobots/include/pinocchio/multibody/model.hxx:30
#1  0x000055555556f076 in __gnu_cxx::__ops::_Iter_pred<pinocchio::details::FilterFrame>::operator()<__gnu_cxx::__normal_iterator<pinocchio::FrameTpl<double, 0> const*, std::vector<pinocchio::FrameTpl<double, 0>, Eigen::aligned_allocator<pinocchio::FrameTpl<double, 0> > > > > (this=0x7fffffffccb0, __it=<error reading variable: Cannot access memory at address 0x5555556a5050>)
    at /usr/include/c++/9/bits/predefined_ops.h:283
#2  0x000055555556d17c in std::__find_if<__gnu_cxx::__normal_iterator<pinocchio::FrameTpl<double, 0> const*, std::vector<pinocchio::FrameTpl<double, 0>, Eigen::aligned_allocator<pinocchio::FrameTpl<double, 0> > > >, __gnu_cxx::__ops::_Iter_pred<pinocchio::details::FilterFrame> > (__first=<error reading variable: Cannot access memory at address 0x5555556a5050>, __last=
                {name = "", parent = 0, previousFrame = 0, placement = {<pinocchio::SE3Base<pinocchio::SE3Tpl<double, 0> >> = {<No data fields>}, rot = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {0, 0, 0, 0, 0, 0, 0, 0, 6.9533480164877528e-310}}}}, <No data fields>}, trans = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {2.1219957904712067e-314, 0, 0}}}}, <No data fields>}}, type = (unknown: 1432950240), inertia = {<pinocchio::InertiaBase<pinocchio::InertiaTpl<double, 0> >> = {<No data fields>}, m_mass = 0, m_com = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {4.6355706025130063e-310, 0, 0}}}}, <No data fields>}, m_inertia = {m_data = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 6, 1, 0, 6, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 6, 1, 0, 6, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 6, 1, 0, 6, 1> >> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 6, 1, 0, 6, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 6, 1, 0, 6, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<d [... cut ...]
#8  0x00007ffff7222afd in jiminy::Robot::initialize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool const&, std::vector<std::__cxx11::basic_string<char, std::char_tra
its<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) ()
   from /home/victor/.local/lib/python3.8/site-packages/jiminy_py/core/lib/libjiminy.so.1.6
#9  0x00005555555619c9 in main (argc=1, argv=0x7fffffffda48) at /home/victor/Documents/atalante_cpp_jiminy/double_pendulum.cc:86

```

Thanks!

[double_pendulum.txt](https://github.com/duburcqa/jiminy/files/6517895/double_pendulum.txt)

vcparedesc, 2021-05-20:
Compiling jiminy from source and compiling the cpp project against the local libraries works. The problem only happens when installing jiminy_py and pinocchio with pip and apt install, respectively. Maybe it is a problem with the installed pinocchio version?

vcparedesc, 2021-05-20:
I installed pinocchio following the instructions in: https://stack-of-tasks.github.io/pinocchio/download.html
with:
 `sudo apt install -qqy robotpkg-py38-pinocchio`

which unpacks:
`robotpkg-py38-pinocchio_2.6.0_amd64.deb`

duburcqa, 2021-05-20:
Ok I see. Jiminy is build against pinocchio 2.5.6, so maybe it is the issue, they did quite a lot of changes in `Frame` recently. For now I should enforce `pinocchio < 2.6.0` in the provided Cmake configuration file of Jiminy installed via `pip`. Could you check what's happen if you downgrade your release of `pinocchio` ? The version of the dependencies used to precompile jiminy is provided [here](https://github.com/duburcqa/jiminy/blob/master/build_tools/easy_install_deps_ubuntu.sh)

```bash
sudo apt-get install -y --allow-downgrades --allow-unauthenticated \
        robotpkg-urdfdom-headers=1.0.4 robotpkg-hpp-fcl=1.7.1 robotpkg-pinocchio=2.5.6 \
        robotpkg-py3*-qt5-gepetto-viewer=4.10.1r2 robotpkg-py3*-qt5-gepetto-viewer-corba=5.5.1 robotpkg-py3*-omniorbpy=4.2.4 \
        robotpkg-py3*-eigenpy=2.6.2 robotpkg-py3*-hpp-fcl=1.7.1 robotpkg-py3*-pinocchio=2.5.6
```

vcparedesc, 2021-05-20:
Some of the packages were having dependencies issues, just as a quick check I installed a subset of your lines:
```
sudo apt-get install -y --allow-downgrades --allow-unauthenticated \
        robotpkg-urdfdom-headers=1.0.4 robotpkg-hpp-fcl=1.7.1 robotpkg-pinocchio=2.5.6 \
        robotpkg-py38-omniorbpy=4.2.4 \                                                                                       
        robotpkg-py38-eigenpy=2.6.2 robotpkg-py38-hpp-fcl=1.7.1 robotpkg-py38-pinocchio=2.5.6
```
... and that solved the problem! Thank you!

closed 2021-05-20 by vcparedesc