duburcqa/jiminy #354: [venv/python dependencies] issues building jiminy with a virtual environment

o-Oscar, 2021-06-17:
Hi,

I am trying to build jiminy with a virtual env : essentially just an other python located on an other folder. 
I was able to build everything, jiminy is accessible to python on the venv, but eigenpy and pinocchio were linked to the wrong python (the system-wide python). 

Thus, when I try to import jiminy from the venv python, I get the following : 

'''
(venv) oscar@oscar-Aspire-A515-54G:~/workspace/jiminy$ python3 -c "import jiminy_py"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/oscar/workspace/jiminy/venv/lib/python3.8/site-packages/jiminy_py/__init__.py", line 1, in <module>
    from .core import *  # noqa: F403
  File "/home/oscar/workspace/jiminy/venv/lib/python3.8/site-packages/jiminy_py/core/__init__.py", line 43, in <module>
    _module = _importlib.import_module(".".join((__name__, module_name)))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'jiminy_py.core.eigenpy'
'''

I guess jiminy was not build to support venv. But I would like to know where I can choose the linked python for eigenpy, pinocchio (and other I might have missed).

o-Oscar, 2021-06-17:
Ok, I found two things that had to be done in order to get my venv to work.

First I had to change the `./build_tools/easy_install_deps_ubuntu.sh` file to account for my setup :
I have to add the following line to activate the environment within the bash file : `source venv/bin/activate`
And change every `sudo -u $(id -nu "$SUDO_UID") python3 -m pip...` to `sudo -u $(id -nu "$SUDO_UID") env "PATH=$PATH" python3 -m pip...` in order to keep the right PATH for the sudo user.

Finally, I had to export a new field in `PYTHONPATH`. Namely : `export PYTHONPATH=/opt/openrobots/lib/python3.8/site-packages:$PYTHONPATH`

With those changes, I can effectively use jiminy_py from the python of my venv !!!

duburcqa, 2021-06-18:
Actually the easy-install is not very well suited for virtual env, since it is used to install system-wide dependencies, which are supposed to work with the native environment of the system. In this particular setupi would rather recommend to build dependencies from source to avoid any mistake, unless we can really find a way to support both native and virtual env in easy-install script out-of-the-box.

duburcqa, 2021-06-18:
> activate the environment within the bash file 

Is this really necessary ? Outside is not enough ?

closed 2021-06-24 by duburcqa

duburcqa, 2021-07-19:
PR https://github.com/duburcqa/jiminy/pull/383 fixing support of venv completely ongoing.