duburcqa/jiminy #391: jiminyConfig.cmake recognizing 2.7 instead of specific python3 version

vcparedesc, 2021-08-05:
Hello,

We are compiling the example in pip_extension on a machine with Ubuntu 16.04 that has several python versions (2.7,  3.6 and 3.8). Whenever we try to run cmake we get that the detected Python version is 2.7 
[image]

When we modify "python" to "python3" in the file jiminyConfig.cmake, we are able to compile the example, with detected python version 3.8.

However, when we use update-alternatives to set the active python version to 3.6, jiminyConfig.cmake still detects 3.8 with the modification, or 2.7 otherwise. 

We need to use python3.6 because pinocchio is provided for py36 for Ubuntu 16.04. Is there some modification we can do to jiminyConfig.cmake to specify the python version or location we need? (3.6)

Thanks :)

duburcqa, 2021-08-05:
Try to specify `PYTHON_EXECUTABLE` in cmake command (or `PYTHON_REQUIRED_VERSION` if you have cmake >= 3.12.4 but I doubt it):

```
cmake "$RootDir/examples/pip_extension" -DCMAKE_INSTALL_PREFIX="$InstallDir" \
            -DPYTHON_EXECUTABLE="/usr/bin/python3.6" -DCMAKE_BUILD_TYPE="Release"
```

vcparedesc, 2021-08-11:
We were using update-alternatives, but somehow it got the python versions mixed up. Using -DPYTHON_EXECUTABLE="/usr/bin/python3.6" solved the problem and we could compile the code. Thanks!

closed 2021-08-11 by vcparedesc