duburcqa/jiminy #403: Meshcat in COLAB

jcarpent, 2021-08-24:
Dear @duburcqa,

Following the discussion https://github.com/RobotLocomotion/drakae/issues/12645, did you manage to get a colab session with MeshCat? If so, could you provide some pointers?
We (with @nim65s) would be interested to watch and take inspiration from the solution.

duburcqa, 2021-08-24:
Hi @jcarpent,

Yes I was able to make it "work", in a sense that it is functional. I'm sending/receiving messages to the javascript viewer in colab directly through the web/zmq sockets of the ipython kernel. For now the solution is hacky because I'm using a very low level and undocumented communication protocol that can be changed from one version of ipython to another (and it does unfortunately). So far I'm sending so many messages that the kernel blocks them at some point and the whole colab session freezes. On a local machine it can be avoided by setting the iopub maximum rate, but it is not possible on colab as far as I know. Also, this solution does not work for some Jupiter interfaces, as the one embedded we VSCode at some point, but I think it works now since they reworked their implementation to be closer to a usual local jupyter session.

traversaro, 2021-08-24:
@jcarpent for the Python part we have something that took inspiration from drake and pinocchio in https://github.com/robotology/idyntree/pull/905, thanks to @GiulioRomualdi .

duburcqa, 2021-08-24:
So the points to improve are:
- the portability (being compatible with any, or at least most ipython versions)
- deal with the iopub rate issue

duburcqa, 2021-08-24:
What I want to emphasize, is that contrary to any existing solution, I'm really using the existing ipython kernel sockets for communicating with the web UI. As a result, there is nothing special to setup, no firewall issue, no ssh issue, no port issue or VPN issue whatsoever. If you can access a jupyter session, it is guaranteed to work out the box whatever the setup. But the cost to pay is to comply with the limitations and restrictions that applies to those sockets.

jcarpent, 2021-08-24:
> What I want to emphasize, is that contrary to any existing solution, I'm really using the existing ipython kernel sockets for communicating with the web UI. As a result, there is nothing special to setup, no firewall issue, no ssh issue, no port issue or VPN issue whatsoever. If you can access a jupyter session, it is guaranteed to work out the box whatever the environment.
> 
> But the cost to pay is to comply with the (undocumented) limitations and restrictions that applies to those sockets.

Could you provide quick guidance to reproduce it?