duburcqa/jiminy #546: Adding position constraint

vcparedesc, 2022-07-01:
Hello,

I am trying to simulate a robot sitting on a chair by constraining a point in the robot to have its position fixed. Can I do this in Jiminy (C++) ?

Thanks!

duburcqa, 2022-07-01:
Hello,

Sure, if it is really what you want, just add a [FixedFrameConstraint](https://github.com/duburcqa/jiminy/blob/686977c798234cf4ed87907a76ca670798e1c36d/core/include/jiminy/core/constraints/FixedFrameConstraint.h)!

Nota Bene: I'm not sure it is the best approach, you could add a "ground" profile having the shape of a chair, and adding contact frame on the frames that are supposed to be in contact.

vcparedesc, 2022-07-02:
Thank you! I think I'll try the ground profile first since it seems more natural. I see I can follow the example in `box_uneven_ground_impulse_contact`, but do you have some documentation or directions for the hyper-parameters of the ground profile?

duburcqa, 2022-07-02:
In the near future, it will be possible to add objects such as a chair on the scene, but it is not possible for now. Same for mesh ground profile instead of implicit function. In the meantime, the ground profile must be a function taking x,y query coordinates as a vector3 (z is irrelevant here) and return a pair with the height and normal at this location.

vcparedesc, 2022-07-14:
I followed your advice and generated the following code, however I'm unable to visualize my new ground profile.

For C++ I added the following test profile:

```
  heightmapFunctor_t custom_ground_profile = [](vector3_t const & pos) -> std::pair<float64_t, vector3_t> {
    if(pos(0) > 0.3) {
      return {1.0,  vector3_t::UnitZ()};
    }else{
      return {0.0,  vector3_t::UnitZ()};
    }
  };
  
  boost::get<heightmapFunctor_t>(boost::get<configHolder_t>(simuOptions.at("world")).at("groundProfile")) = custom_ground_profile;
```

And, I wanted to visualize this in python, but couldn't figure out a way to use the HeightMapFunctor, what I tried is:
```

from jiminy_py.core import HeightmapFunctor

custom_ground_profile = lambda  position : (0, np.array([0, 0, 1])) if position[0] < 0.3 else (1, np.array([0, 0, 1]))
engine_options["world"]["groundProfile"] = HeightmapFunctor([custom_ground_profile])
engine.set_options(engine_options)
```

But that's not showing me a change in terrain when i call `play_logs_data`. Could you point me in the right direction?

Thanks

vcparedesc, 2022-07-14:
Thanks for the help!

I think the c++ ground profile works as the robot seats and keeps stability in place. 
[image]

I tested using the ground profile of the `box_uneven_ground_impulse_contact.py` and I can visualize it, so using a viewer and forwarding it to `play_logs_data` is working for me.

However, since I can't visualize my custom ground profile, I think I'm doing something wrong writing the python call to he HeightMapFunction, I am doing this:

```
custom_ground_profile = lambda self,  px, py, pz : (0, np.array([0, 0, 1])) if px < 0.3 else (1, np.array([0, 0, 1]))
hmap = HeightmapFunctor(custom_ground_profile)

engine_options["world"]["groundProfile"] = hmap
engine.set_options(engine_options)
```
Do you have any suggestion to fix it?

Thanks

duburcqa, 2022-07-15:
Have you tried what I suggest with `update_floor` ? Do you have a more complete snippet to understand what is going wrong ?

duburcqa, 2022-07-15:
The signature of your ground profile in Python is wrong. It should be:
```
def custom_ground_profile(px, py, height, normal) :
    if px < 0.3:
        height[()] = 1.0
    else:
        height[()] = 0.0
    normal[:] = 0.0, 0.0, 1.0
```

vcparedesc, 2022-07-15:
I changed the ground profile as you suggested, and I can see my robot sitting :)

[image]

I did use your suggestions for 'update_floor'

```
from jiminy_py.viewer import Viewer

viewer = Viewer(robot)

viewer.update_floor(ground_profile, show_meshes=False)
play_logs_data(robot, log_data,
               speed_ratio=0.5,
               viewers=viewer,
               camera_xyzrpy=[(0.0, 7.0, 0.0), (np.pi/2, 0.0, np.pi)],
               delete_robot_on_close=False)
```

Thanks for the help.

closed 2022-07-15 by vcparedesc

duburcqa, 2022-07-15:
Nice ! Perfect. Are you using the latest release ? Because the rendering looks quite ugly.

vcparedesc, 2022-07-15:
I'm using Jiminy 1.7.4, I might update to the latest release, but for now this version was stable enough without investigating if some functions have changed wrt to newer versions.

duburcqa, 2022-07-15:
Ok yes I understand then. I improved the rendering quite recently. And yes the API is mostly stable.