duburcqa/jiminy #560: Failures in engine while trying to run Environment

Le5tes, 2022-08-17:
Hi, 
I'm trying to train an agent to run the Anymal Environment in jupyter notebook with python 3.8.10.

I'm getting this error in the notebook:
```
RuntimeError                              Traceback (most recent call last)
<ipython-input-58-238be5bf9bbb> in <module>

---> 18     env.step(action)

~/.local/lib/python3.8/site-packages/gym_jiminy/common/envs/env_generic.py in step(self, action)
    745 
    746         # Perform a single simulation step
--> 747         self.simulator.step(self.step_dt)
    748 
    749         # Update shared buffers

~/.local/lib/python3.8/site-packages/jiminy_py/simulator.py in step(self, step_dt)
    404         return_code = self.engine.step(step_dt)
    405         if return_code != jiminy.hresult_t.SUCCESS:
--> 406             raise RuntimeError("Failed to perform the simulation step.")
    407 
    408     def simulate(self,

RuntimeError: Failed to perform the simulation step.
```

And I'm getting this error printed to the terminal:
```
In /__w/jiminy/jiminy/core/src/engine/EngineMultiRobot.cc:2242: In jiminy::EngineMultiRobot::step:
error: Too many successive iteration failures. Probably something is going wrong with the physics. Aborting integration.
```

Simple code to reproduce (you may have to run a few times)
```
from gym_jiminy.envs import ANYmalJiminyEnv

env = ANYmalJiminyEnv()

env.reset()

for i in range(1000):
    action = env.action_space.sample()
    env.step(action)
  

print('do I get here?')

```

duburcqa, 2022-08-17:
Indeed there is an issue. I think it comes from the contact model that is based by default on collision geometries instead of frames. The former is supposed to be more realistic but untested for now and numerically unstable model. I'm going to rework internally this part of the engine in a near future to handle generic collisions (collisions of all bodies with the ground plus auto-collisions).

duburcqa, 2022-08-18:
I see you did some work based on jiminy [here](https://github.com/Le5tes/robot-reinforcement-learning). If you have any question or want me to give you some advice / code review, do not hesitate. The documentation of jiminy is quite poor and providing a comprehensive tutorial would be amazing !

Le5tes, 2022-08-18:
Thanks!
Yes, it's an assignment for a Master's. Once we're finished I'm happy to make a tutorial from it. :)

duburcqa, 2022-08-18:
So, in the end do you need me to release a new version fixing the issue or you are planning to overwrite the base configuration manually?

Le5tes, 2022-08-18:
I overwrote the base configuration in my downloaded version of the package, so no worries!

duburcqa, 2022-08-18:
Perfect !

duburcqa, 2022-08-19:
There is one PID controller per motor. Each of them are responsible for the sending actual torque based on the error between the current and target state for the motors. More details are available [here](https://en.wikipedia.org/wiki/PID_controller). In this case, the action is no longer the torque directly, but the target state of each motor. These low-level PID controllers are supposed to update the current torque at a higher rate than the action. More information about why doing this is important can be found in the sub-section about the action space of my preprint [available on arxiv](https://arxiv.org/pdf/2203.01148.pdf). 

For simplicity, I advice you to only ignore completely the target velocity and set it to zero systematically, so that the policy only output the desired position of the motors and not the combination of both position and velocity.

SDJoeKing, 2022-08-19:
Thank you very much for your swiftly response! I will give that a try.

closed 2022-08-27 by duburcqa