duburcqa/jiminy #594: [jiminy/log] More effective way to restore model.

duburcqa, 2023-03-06:
It would be more efficient and clearer to loop over all the attributes of `__dict__` or `vars` instead of leveraging the serialization capability of pinocchio itself:
```python
for name, value in vars(pinocchio_model).items():
    robot.pinocchio_model[name] = value
for name, value in vars(pinocchio_data).items():
    robot.pinocchio_data[name] = value
```

https://github.com/duburcqa/jiminy/blob/809ecf0a2e9d2cf7984cc7937ab4f269bb289327/python/jiminy_py/src/jiminy_py/log.py#LL204C9-L204C48

jcarpent, 2023-03-07:
I don’t think this solution will be faster than internal Pinocchio serialization

duburcqa, 2023-03-07:
In the end I'm going for:
```
robot.pinocchio_model.__setstate__(pinocchio_model.__getstate__())
pinocchio_data = pin.Data(pinocchio_model)
robot.pinocchio_data.__setstate__(pinocchio_data.__getstate__())
```

jcarpent, 2023-03-07:
So using Pinocchio serialization

duburcqa, 2023-03-07:
yes, It makes more sense like this, because relying on `__dict__` turns out to be hazardous because some attributes may be read-only.

closed 2023-03-11 by duburcqa