duburcqa/jiminy #598: Feature Request: migrate from gym to gymnasium

elliottower, 2023-03-17:
Hi, would it be possible for jiminy to be upgraded from gym to gymnasium? Gymnasium is the maintained version of openai gym and is compatible with current RL training libraries ([rllib](https://github.com/ray-project/ray/tree/master/rllib) and [tianshou](https://github.com/thu-ml/tianshou) have already migrated, and [stable-baselines3](https://github.com/DLR-RM/stable-baselines3/blob/e5deeed16efb57c34ccdcb14692439154d970527/docs/guide/install.rst#bleeding-edge-version) will soon).

This repository is currently listed in the gymnasium [third party environments](https://gymnasium.farama.org/environments/third_party_environments/) but we are cleaning the list up to only include maintained gymnasium-compatible repositories.

For information about upgrading and compatibility, see [migration guide](https://gymnasium.farama.org/content/migration-guide/) and [gym compatibility](https://gymnasium.farama.org/content/gym_compatibility/).

duburcqa, 2023-03-18:
I want to migrate to gymnasium but i'm wanting for stable baselines3 to release à compatible version. For now, gym is the only version that is compatible with all the frameworks you mentioned. Regarding gymnasium there is already an issue since the version is pinned to 0.26.3 while it is pinned to 0.26.2 by the future release of stable-baselines3...

elliottower, 2023-03-21:
> I want to migrate to gymnasium but i'm wanting for stable baselines3 to release a compatible version. For now, gym is the only version that is compatible with all the frameworks you mentioned. Regarding gymnasium there is already an issue since the version is pinned to 0.26.3 while it is pinned to 0.26.2 by the future release of stable-baselines3...
> 
> But for sure jiminy is actively maintained and most use-cases imply gym/gymnasium.

I don’t believe sb3’s gymnasium compatibility is pinned to a specific version, if you look through the files changes by the PR, it doesn’t seem specific to any version. I’ve tested it in my own use cases and found no issues, not sure about the timeline for it getting merged though as it’s a draft currently and has been for a while. https://github.com/DLR-RM/stable-baselines3/pull/1327/files

But makes sense to wait until things are officially compatible and simply use older versions of the libraries and gym until then, just figured I’d post in case it wasn’t on your radar.

duburcqa, 2023-03-21:
After checking again, current it is pinned to the `main` branch. Which is not any better than before. In `setup.py`: 
```
"gymnasium @ git+https://github.com/Farama-Foundation/Gymnasium@main",
```