duburcqa/jiminy #668: Cannot load robot based on urdf and example is broken on MacOS M2 due to missing gym-jiminy

sebbyjp, 2023-12-02:
Hello, I love the idea for this library! Pip cannot find `jiminy_py` on arm64-based docker Ubuntu 22.04 python3.10 so I am trying to run it on my mac:

MacOS: Sonoma
Python: 3.11.6

# Attempt to load robot in simulator:
```
import os
import numpy as np

import jiminy_py.core as jiminy  # The main module of jiminy - this is what gives access to the Robot
from jiminy_py.simulator import Simulator

try:
    from importlib.resources import files
except ImportError:
    from importlib_resources import files

mesh_dirs =["/Users/sebastianperalta/simply/dev/simply-mono/interbotix/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions/meshes/",
	"simply-mono/interbotix/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions/meshes/locobot_meshes/",
	"simply-mono/interbotix/interbotix_ros_rovers/interbotix_ros_xslocobots/interbotix_xslocobot_descriptions/meshes/locobot_meshes/mobile_wx250s_meshes/",
	"simply-mono/interbotix/interbotix_ros_rovers/interbotix_ros_xslocobots/"]
urdf_path = "/Users/sebastianperalta/simply/dev/simply-mono/dgl_ros/loco.urdf"

# Instantiate and initialize the robot
robot = jiminy.Robot()
robot.initialize(urdf_path, mesh_package_dirs=mesh_dirs)

# Add a single motor
# motor = jiminy.SimpleMotor("PendulumJoint")
# robot.attach_motor(motor)
# motor.initialize("PendulumJoint")

# Define the command: for now, the motor is off and doesn't modify the output torque.
def compute_command(t, q, v, sensors_data, command):
    command[:] = 0.0

# Instantiate and initialize the controller
# controller = jiminy.ControllerFunctor()
# controller.initialize(robot)

# Create a simulator using this robot and controller
simulator = Simulator(robot)

# Set initial condition and simulation length
q0, v0 = np.array([0.1]), np.array([0.0])
simulation_duration = 10.0

# Launch the simulation
simulator.simulate(simulation_duration, q0, v0)
```

Error:
[image]

# Attempt to run example on your docs website:
```
import os

import numpy as np

import jiminy_py.core as jiminy  # The main module of jiminy - this is what gives access to the Robot
from jiminy_py.simulator import Simulator

try:
    from importlib.resources import files
except ImportError:
    from importlib_resources import files

data_root_path = files("gym_jiminy.envs") / "data/toys_models/simple_pendulum"
urdf_path = str(data_root_path / "simple_pendulum.urdf")

# Instantiate and initialize the robot
robot = jiminy.Robot()
robot.initialize(urdf_path, mesh_package_dirs=[data_root_path])

# Add a single motor
motor = jiminy.SimpleMotor("PendulumJoint")
robot.attach_motor(motor)
motor.initialize("PendulumJoint")

# Define the command: for now, the motor is off and doesn't modify the output torque.
def compute_command(t, q, v, sensors_data, command):
    command[:] = 0.0

# Instantiate and initialize the controller
controller = jiminy.ControllerFunctor()
controller.initialize(robot)

# Create a simulator using this robot and controller
simulator = Simulator(robot, controller)

# Set initial condition and simulation length
q0, v0 = np.array([0.1]), np.array([0.0])
simulation_duration = 10.0

# Launch the simulation
simulator.simulate(simulation_duration, q0, v0)
```
Error:
[image]

Also, why do I get this "Python quit unexpectedly pop-up"? No other python program has produced something like that in my recent memory.

Anyways, thanks again for making tools like this!

duburcqa, 2023-12-03:
> Pip cannot find jiminy_py on arm64-based docker Ubuntu 22.04 python3.10

Indeed, this is expected. I'm not providing precompiled wheels for ARM architecture for now, because github does not provide such machines and doing cross-compilation on x86 would require some effort to setup (not clearly not impossible). If there is a real need I will do it at some point in the future for sure.

Could you provide the complete trace for the exception? Instead of a screenshot ? It would be much easier for me to help you. Unfortunately I have no spare Mac M2  on which I could investigate the issue myself. Jiminy has only been thoroughly tested on X86 CPU architecture so I'm not very surprise there is still issue to fix on Mac M2 for now.

sebbyjp, 2023-12-04:
Sure thing, here is the trace with the example from the docs:
```
(venv) sebastianperalta@Sebastians-MacBook-Pro dev % python jim_example.py 
HDF5-DIAG: Error detected in HDF5 (1.14.0) thread 0:
  #000: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
  #001: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2540 in H5T__register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
  #002: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 5132 in H5T__path_find_real(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
  #003: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7343 in H5T__conv_ushort_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
  #004: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
  #005: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2540 in H5T__register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
  #006: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 5132 in H5T__path_find_real(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
  #007: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7283 in H5T__conv_short_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
  #008: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
  #009: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2540 in H5T__register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
  #010: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 5132 in H5T__path_find_real(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
  #011: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7223 in H5T__conv_uchar_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
  #012: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
  #013: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2540 in H5T__register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
  #014: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 5132 in H5T__path_find_real(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
  #015: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7163 in H5T__conv_schar_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
  #016: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
  #017: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2540 in H5T__register(): unable to locate/allocate conversion path
    major: Datatype
    minor: Unable to initialize object
  #018: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 5132 in H5T__path_find_real(): unable to initialize conversion function
    major: Datatype
    minor: Unable to initialize object
  #019: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7103 in H5T__conv_ldouble_double(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
  #020: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5T.c line 2491 in H5T__register_int(): unable to register internal datatype conversion routine
    major: Datatype
    minor: Unable to initialize object
[... cut ...]
  #031: /Users/runner/work/jiminy/jiminy/build/soup/hdf5/hdf5_external-prefix/src/hdf5_external/src/H5Tconv.c line 7023 in H5T__conv_float_ldouble(): disagreement about datatype size
    major: Datatype
    minor: Unable to initialize object
Traceback (most recent call last):
  File "/Users/sebastianperalta/simply/dev/jim_example.py", line 19, in <module>
    robot.initialize(urdf_path, mesh_package_dirs=[data_root_path])
TypeError: No registered converter was able to produce a C++ rvalue of type std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> from this Python object of type PosixPath
zsh: segmentation fault  python jim_example.py
```

duburcqa, 2023-12-05:
OK the HDF5 error is strange, I will try to have a look at it. Regarding the segmentation fault at the end, it is due to either `urdf_path` or `data_root_path` not being a plain string but a `Path` object. Try doing this instead:
```
robot.initialize(str(urdf_path), mesh_package_dirs=[str(data_root_path)])
```
Of course it is a big that should be fixed at jiminy level.

duburcqa, 2023-12-05:
I can confirm that HDF5 errors are related to cross-compilation issues (mentioned [here](https://github.com/PyTables/PyTables/issues/887#issuecomment-1068166949)). I was not aware of such limitation. I will see if I can fix it.

duburcqa, 2023-12-05:
White I'm fixing the issue with HDF5, you can change the log format to "binary" (with ".data") extension for log files. It has the same features but it can only be read usine tools provided by jiminy.

duburcqa, 2023-12-13:
New wheels for MacOS are available [here](https://github.com/duburcqa/jiminy/actions/runs/7193017600). You can check if it fixes your issues.

closed 2023-12-14 by duburcqa

duburcqa, 2023-12-19:
I now have a MacBook M3 and I can confirm the latest wheel is working like a charm !