duburcqa/jiminy #691: How to add collision object to the scene?

fedor-chervinskii, 2024-01-12:
Hi @duburcqa! This is a brilliant tool and I like how easy it is to start using, and that it's working headless and RL-ready.

However, for my use case, I'd like to have some objects that are external to the robot, and also potentially collidable. I couldn't find anything relevant in the documentation. What's the best/easiest way to do that in Jiminy? If it requires some patching - maybe if you could explain it, I could try to implement it myself.

An example would be a pick-and-place task for a robot.

Thanks!

duburcqa, 2024-01-15:
Hi @fedor-chervinskii ,

Thank you for your support ! Unfortunately, this feature is not available at the time being. It is part of the highly requested features that I'm going to implement ASAP, but it may take a while nonetheless. I'm not expecting this before at least  3 months because I don't have time to work on it for now. [Here](https://github.com/duburcqa/jiminy/discussions/508) is the list of all the features that are going to be implemented at some point.

For now, only collision with the ground is supported, and the ground has to be a height map. I know it is very limiting...

duburcqa, 2024-01-15:
If you just want to avoid colliding with the environment, you could create a height map that includes big "bumps" here and there as obstacle. As such, those obstacles will be weightless and unmovable. For now there is no better way.

The necessary refactoring to support generic collisions is quite significant and must be implemented in the C++ core. First, `hpp-fcl` must be leveraged to detect the collision between bodies. There is already a draft implementation of this merged in jiminy itself but it does not work reliably often for now. Then, additional collision pairs between objects should be added to not consider the ground only.

At python level, it is still possible to detect generic collision but not to grasps stuffs. [here](https://github.com/duburcqa/jiminy/blob/dev/python/jiminy_py/examples/collision_detection.py) is an example for plain collision detection in Python.

closed 2024-01-25 by duburcqa