duburcqa/jiminy #771: `build_robot_from_log` no longer works due to backlash

matthieuvigne, 2024-04-22:
When loading a non-persistent log with `build_robot_from_log`, then looking at `robot.log_position_fieldnames`, backlash joints are missing from this list. This in turn breaks replay functions - that rely on this variable through `extract_trajectory_from_log`.

The root cause seems to be the fact that, when creating the robot, the hardware and specifically the motors are not included - and  backlash is a motor option. Thus, when doing `robot.set_options(all_options[robot_name or "robot"])`, since there are no motors, the motors are not added.

Note that because `pinocchio_model` is in the log, this object is correct. This in fact leads to the mismatch: if it were recreated from the options, it too would be missing the backlash (which would be wrong but consistent).

I see two ways to handle the problem:

 - try to rebuild the hardware in `build_robot_from_log`, thus restoring the backlash
 - a more ad-hoc fix, use `pinocchio_model` to regenerate the list `log_position_fieldnames`

I prefer the first solution because it seems more robust. However, I don't think there is a way to know the *type* of motor that was added from the log. Here is a proposed fixed, with works in my context but assumes that the motor is of type `jiminy.SimpleMotor`:

```diff
diff --git a/python/jiminy_py/src/jiminy_py/log.py b/python/jiminy_py/src/jiminy_py/log.py
index b4240414..6d9699a2 100644
--- a/python/jiminy_py/src/jiminy_py/log.py
+++ b/python/jiminy_py/src/jiminy_py/log.py
@@ -214,6 +214,12 @@ def build_robot_from_log(
 
         # Load the options
         all_options = log_constants["options"]
+        # Add a SimpleMotor to the joint that had a motor
+        motor_options = all_options[robot_name or "robot"]["motors"]
+        for joint_name in motor_options.keys():
+            motor = jiminy.SimpleMotor(joint_name)
+            robot.attach_motor(motor)
+            motor.initialize(joint_name)
         robot.set_options(all_options[robot_name or "robot"])
 
         # Update model in-place.
```

duburcqa, 2024-04-22:
Being able to re-create robots from log properly is a long overdue issue. Currently, we pretend that the extended model is actually the theoretical one when instantiating the robot. Because of this, it is not possible to restore the options properly, and it cannot be used for anything else than replay since everything related to the difference between theoretical and extended model is broken...

The missing part to fix this issue is to move the notion of hardware description from Python to C++, so that it can be exported and loaded. Once this is done, it should be straightforward to build the original model and load hardware and options to get the right extended model.

I will have a look at it ASAP, it has been annoying me since a long time already.

matthieuvigne, 2024-04-23:
All right. How long do you think it will take you ? I don't want to rush you so I'm thinking it might be best to patch this in my own fork in an intermediate release while you work on a more long-term solution.

duburcqa, 2024-04-23:
I think this is definitely the right approach, no matter how long it takes on my side, unless you can open the MR yourself. For this particular matter, it is hard to tell. I'm almost done backporting all the logic for loading a `jiminy::Model` from Python to C++.  Next I need to do the same for `jiminy::Robot`, which basically means adding hardware. If everthing goes well, it could be today or tomorrow.

duburcqa, 2024-04-24:
OK I'm done with the serialisation of 'Model'. Now it remains to deal with all the additions of a robot ! It should not be too hard now that the architecture is there :)

closed 2024-04-28 by duburcqa